VIEWESMART/UEDX46460015-MD50ESP32-1.5inch-Touch-Knob-Display #2: User data should be placed in SRAM

par-n-go, 2025-03-02:
Following your instructions on how to port and upload the code through the arduino IDE I get this error while trying to compile an example:

```
c:\Projects\screen_test\libraries\ESP32_Display_Panel-bugfix-missing_lcd_load_vendor_config\src\lcd\ESP_PanelLcd.cpp:457:109: error: macro "ESP_PANEL_CHECK_FALSE_RET" requires 4 arguments, but only 1 given
  457 |         ESP_PANEL_CHECK_FALSE_RET((esp_ptr_internal(user_data), false, "User data should be placed in SRAM"));
      |                                                                                                             ^
In file included from c:\Projects\screen_test\libraries\ESP32_Display_Panel-bugfix-missing_lcd_load_vendor_config\src\lcd\ESP_PanelLcd.cpp:9:
c:\Projects\screen_test\libraries\ESP32_Display_Panel-bugfix-missing_lcd_load_vendor_config\src/ESP_PanelLog.h:58: note: macro "ESP_PANEL_CHECK_FALSE_RET" defined here
   58 | #define ESP_PANEL_CHECK_FALSE_RET(x, ret, fmt, ...) do {         \
      | 
c:\Projects\screen_test\libraries\ESP32_Display_Panel-bugfix-missing_lcd_load_vendor_config\src\lcd\ESP_PanelLcd.cpp: In member function 'bool ESP_PanelLcd::attachRefreshFinishCallback(std::function<bool(void*)>, void*)':
c:\Projects\screen_test\libraries\ESP32_Display_Panel-bugfix-missing_lcd_load_vendor_config\src\lcd\ESP_PanelLcd.cpp:454:29: error: cannot convert 'std::function<bool(void*)>' to 'const void*'
  454 |             esp_ptr_in_iram(callback), false,
      |                             ^~~~~~~~
      |                             |
      |                             std::function<bool(void*)>
c:\Projects\screen_test\libraries\ESP32_Display_Panel-bugfix-missing_lcd_load_vendor_config\src/ESP_PanelLog.h:59:14: note: in definition of macro 'ESP_PANEL_CHECK_FALSE_RET'
   59 |         if ((x) == false) {                                      \
      |              ^
In file included from C:\Users\valer\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-489d7a2b-v1\esp32s3/include/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:54,
                 from C:\Users\valer\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-489d7a2b-v1\esp32s3/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:57,
                 from C:\Users\valer\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-489d7a2b-v1\esp32s3/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:69,
                 from C:\Users\valer\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-489d7a2b-v1\esp32s3/include/esp_driver_spi/include/driver/spi_master.h:10,
                 from C:\Users\valer\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-489d7a2b-v1\esp32s3/include/esp_lcd/include/esp_lcd_io_spi.h:11,
                 from C:\Users\valer\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-489d7a2b-v1\esp32s3/include/esp_lcd/include/esp_lcd_panel_io.h:13,
                 from c:\Projects\screen_test\libraries\ESP32_Display_Panel-bugfix-missing_lcd_load_vendor_config\src\lcd\ESP_PanelLcd.cpp:13:
C:\Users\valer\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-489d7a2b-v1\esp32s3/include/esp_hw_support/include/esp_memory_utils.h:41:48: note:   initializing argument 1 of 'bool esp_ptr_in_iram(const void*)'
   41 | inline static bool esp_ptr_in_iram(const void *p) {
      |                                    ~~~~~~~~~~~~^
c:\Projects\screen_test\libraries\ESP32_Display_Panel-bugfix-missing_lcd_load_vendor_config\src\lcd\ESP_PanelLcd.cpp:457:9: error: 'ESP_PANEL_CHECK_FALSE_RET' was not declared in this scope
  457 |         ESP_PANEL_CHECK_FALSE_RET((esp_ptr_internal(user_data), false, "User data should be placed in SRAM"));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

VIEWESMART, 2025-03-03:
> Following your instructions on how to port and upload the code through the arduino IDE I get this error while trying to compile an example:
>
> ``` [...]

Thank you for your feedback. This error is related to the version of the development board you are using. It is recommended that you use version v3.0.6-v3.1.0 for the ESP32 core. Please feel free to provide feedback if you encounter any issues, and I will do my best to help you resolve them.

par-n-go, 2025-03-10:
Thank you, issue is solved

closed 2025-03-10 by par-n-go