VIEWESMART/UEDX46460015-MD50ESP32-1.5inch-Touch-Knob-Display #3: Acccess boot mode

Utstallningsverkstaden, 2025-03-03:
I need to upload a new program in Arduino IDE and it worked the first time, but now it won't upload after the compile is done (serial is working).
How can I access the boot mode pin? Is it possible via the breakout board or do I have to open up the display to reach the esp32 somehow?

VIEWESMART, 2025-03-04:
> I need to upload a new program in Arduino IDE and it worked the first time, but now it won't upload after the compile is done (serial is working). How can I access the boot mode pin? Is it possible via the breakout board or do I have to open up the display to reach the esp32 somehow?

You can try holding down the knob to make it enter the boot mode for burning. Then see if the burning can be successful.

Utstallningsverkstaden, 2025-03-04:
Sadly, that didn't work, nothing happened.

> Sketch uses 975604 bytes (31%) of program storage space. Maximum is 3145728 bytes.
> Global variables use 72600 bytes (22%) of dynamic memory, leaving 255080 bytes for local variables. Maximum is 327680 bytes.
> esptool.py v4.8.1
> Serial port COM13
> Connecting....
> 
> A serial exception error occurred: Write timeout
> Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
> For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
> Failed uploading: uploading error: exit status 1

The first upload i missed to copy the display driver so it wouldn't run, this is the error message that keeps looping:

>  08:42:31.253 -> SPIWP:0xee
> 08:42:31.253 -> mode:DIO, clock div:1
> 08:42:31.253 -> load:0x3fce2820,len:0x1188
> 08:42:31.285 -> load:0x403c8700,len:0x4
> 08:42:31.285 -> load:0x403c8704,len:0xbf0
> 08:42:31.285 -> load:0x403cb700,len:0x30e4
> 08:42:31.285 -> entry 0x403c88ac
> 08:42:31.481 -> [E][Panel][esp_panel_board.cpp:0055](init): 
> 08:42:31.481 -> No default board configuration detected. There are three ways to provide a default configuration: 
> 08:42:31.481 -> 	1. Use the `esp_panel_board_supported_conf.h` file to enable a supported board. 
> 08:42:31.481 -> 	2. Use the `esp_panel_board_custom_conf.h` file to define a custom board. 
> 08:42:31.512 -> 	3. Use menuconfig to enable a supported board or define a custom board. 
> 08:42:31.512 -> [E][Panel][esp_panel_board.cpp:0055](init): 
> 08:42:31.512 -> No default board configuration detected. There are three ways to provide a default configuration: 
> 08:42:31.512 -> 	1. Use the `esp_panel_board_supported_conf.h` file to enable a supported board. 
> 08:42:31.512 -> 	2. Use the `esp_panel_board_custom_conf.h` file to define a custom board. 
> 08:42:31.545 -> 	3. Use menuconfig to enable a supported board or define a custom board. 
> 08:42:31.545 -> [E][Panel][esp_panel_board.cpp:0250](begin): Init failed
> 08:42:31.545 -> [E][LvPort][lvgl_port_v8.cpp:0740](lvgl_port_init): Invalid LCD device
> 08:42:31.545 -> [E][LvPort][lvgl_port_v8.cpp:0812](lvgl_port_lock): LVGL mutex is not initialized
> 08:42:31.577 -> [Warn]	(0.098, +98)	 lv_disp_get_scr_act: no display registered to get its active screen 	(in lv_disp.c line #54)
> 08:42:31.577 -> [Warn]	(0.108, +10)	 lv_disp_get_scr_act: no display registered to get its active screen 	(in lv_disp.c line #54)
> 08:42:31.869 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 08:42:31.869 -> 
> 08:42:31.869 -> Core  1 register dump:
> 08:42:31.869 -> PC      : 0x4200c76b  PS      : 0x00060530  A0      : 0x8200d220  A1      : 0x3fcaf390  
> 08:42:31.869 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000036  A5      : 0x3c06e4ee  
> 08:42:31.869 -> A6      : 0x3c062004  A7      : 0x00000050  A8      : 0x0000006c  A9      : 0x3fc97cf0  
> 08:42:31.869 -> A10     : 0x0000001a  A11     : 0x3c062083  A12     : 0x00000036  A13     : 0x00000014  
> 08:42:31.869 -> A14     : 0x0000000a  A15     : 0x3c06e4ee  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
> 08:42:31.869 -> EXCVADDR: 0x00000020  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  
> 08:42:31.869 -> 
> 08:42:31.869 -> 
> 08:42:31.869 -> Backtrace: 0x4200c768:0x3fcaf390 0x4200d21d:0x3fcaf3c0 0x4200d851:0x3fcaf3e0 0x4200f00a:0x3fcaf410 0x42002614:0x3fcaf430 0x4203018e:0x3fcaf490 0x4037c55a:0x3fcaf4b0
> 08:42:31.869 -> 
> 08:42:31.869 -> 
> 08:42:31.869 -> 
> 08:42:31.869 -> 
> 08:42:31.869 -> ELF file SHA256: 38e42129d
> 08:42:31.869 -> 
> 08:42:31.869 -> Rebooting...

VIEWESMART, 2025-03-04:
> Sadly, that didn't work, nothing happened.
>
>
> The first upload i missed to copy the display driver so it wouldn't run, this is the error message that keeps looping:

Sorry, I want to know if you mean that you can't upload the program or if there are error messages after the program is uploaded.

If you can't burn the program, pressing the button of the knob and then powering on to enter the boot mode for burning as I mentioned before should be feasible. If it doesn't work, we can also use UART for burning.

If the code can be uploaded normally, but the device keeps restarting, I see that the error messages are:
08:42:31.545 -> [E][Panel]esp_panel_board.cpp:0250: Init failed
08:42:31.545 -> [E][LvPort]lvgl_port_v8.cpp:0740: Invalid LCD device
08:42:31.545 -> [E][LvPort]lvgl_port_v8.cpp:0812: LVGL mutex is not initialized
It seems that some files are missing, or the libraries are not quite compatible.

I hope to get more detailed information so that I can help you.

Utstallningsverkstaden, 2025-03-04:
Now I got it to work, when held down knob before powering on, Previously I held it down after the compile was done, when going into upload. Thank you for mentioning that!

VIEWESMART, 2025-03-04:
> Now I got it to work, when held down knob before powering on, Previously I held it down after the compile was done, when going into upload. Thank you for mentioning that!

I'm very glad to hear this news. I'm really sorry for not mentioning this before, which has wasted your time.