VIEWESMART/UEDX46460015-MD50ESP32-1.5inch-Touch-Knob-Display #5: Adapter Board Information

jon68z, 2025-03-23:
Hello, can you please provide information on the adapter board? A schematic and physical drawing would be very much appreciated,

Thanks!

VIEWESMART, 2025-03-24:
> Hello, can you please provide information on the adapter board? A schematic and physical drawing would be very much appreciated,
> 
> Thanks!

I'm very sorry. We haven't created a schematic diagram because its principle is simple, but we will add it later. I can provide you with some information about its interfaces, as shown in the following picture:

[image]

The position of J2 in the picture is the connection location of the flat cable. The internal pins connected are shown in the following picture:

[image]

The corresponding pin information below J2 is shown in the following picture:

[image]

VIEWESMART, 2025-03-24:
> Hello, can you please provide information on the adapter board? A schematic and physical drawing would be very much appreciated,
> 
> Thanks!

I'm very sorry. We haven't created a schematic diagram because its principle is simple, but we will add it later. I can provide you with some information about its interfaces, The physical picture is as follows:

[image]

The position of J2 in the picture is the connection location of the flat cable. The internal pins connected are shown in the following picture:

[image]

The corresponding pin information below J2 is shown in the following picture:

[image]