VIEWESMART/UEDX46460015-MD50ESP32-1.5inch-Touch-Knob-Display #6: Wrong drivers ?

soapyapi, 2025-05-18:
Hello, 

Can you explain how the implementation of the driver works in your code ? Why do you use sh8601 and manually configure it with led arrays instead of of co5300 and a proper qspi setup ? I might work in the example code, but has proven very hard to change/port, especially with newer lvgl releases (9+).

Thank you very much,

VIEWESMART, 2025-05-19:
> Hello,
> 
> Can you explain how the implementation of the driver works in your code ? Why do you use sh8601 and manually configure it instead of of co5300 and a proper qspi setup ? It might work in the example code, but has proven very hard to change/port, especially with newer lvgl releases (9+).
> 
> Thank you very much,

Hello，
Because there is no co5300 driver in the official driver library for the time being, it is used to modify the compatible driver after the initialization instruction instead. It is also QSPI, which does not affect its work. As far as the current example is concerned, it can only be implemented based on lvgl v8, because we are porting v8 in the lvgl_port.

[image]

[image]

qwerkus-1, 2025-05-20:
Thank you for your reply. Actually there is a co5300 driver available here: https://components.espressif.com/components/kodediy/esp_lcd_co5300 nnd it works in lvgl 8.3.
I'm trying to port it to lvgl 9 without success so far. Display turns on but colors are wront a buggy. Can you help me out here?

[image]

Here is my code: 

`#include <stdio.h>
#include <unistd.h>
#include <sys/lock.h>
#include <sys/param.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "driver/spi_master.h"
#include "esp_timer.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "esp_err.h"
#include "esp_log.h"
#include "lvgl.h"
#include "esp_lcd_co5300.h"
#include "ui.h"
#include "lv_examples.h"

#define LCD_HOST               (SPI2_HOST)
#define PIN_NUM_LCD_CS         (GPIO_NUM_12)
#define PIN_NUM_LCD_PCLK       (GPIO_NUM_10)
#define PIN_NUM_LCD_DATA0      (GPIO_NUM_13)
#define PIN_NUM_LCD_DATA1      (GPIO_NUM_11)
#define PIN_NUM_LCD_DATA2      (GPIO_NUM_14)
#define PIN_NUM_LCD_DATA3      (GPIO_NUM_9)
#define PIN_NUM_LCD_RST        (GPIO_NUM_8)
#define PIN_NUM_BK_LIGHT       (GPIO_NUM_17)
#define LCD_H_RES              471
#define LCD_V_RES              466
#define LCD_BIT_PER_PIXEL      16
#define LCD_BK_LIGHT_ON_LEVEL  1

#define LVGL_DRAW_BUF_LINES    20 // number of display lines in each draw buffer
#define LVGL_TICK_PERIOD_MS    2
#define LVGL_TASK_MAX_DELAY_MS 500
#define LVGL_TASK_MIN_DELAY_MS 1000 / CONFIG_FREERTOS_HZ
#define LVGL_TASK_STACK_SIZE   (4 * 1024)
#define LVGL_TASK_PRIORITY     2

static const char *TAG = "main";

// Custom initalizations according to vendor's example - doesn't seem to make a difference
static const co5300_lcd_init_cmd_t lcd_init_cmds[] = {
    {0xFE, (uint8_t []){0x00}, 0, 0},
    {0xC4, (uint8_t []){0x80}, 1, 0},
    {0x3A, (uint8_t []){0x55}, 1, 0},
    {0x35, (uint8_t []){0x00}, 0, 10},
    {0x53, (uint8_t []){0x20}, 1, 10},
    {0x51, (uint8_t []){0xFF}, 1, 10},
    {0x63, (uint8_t []){0xFF}, 1, 10},
    //{0x2A, (uint8_t []){0x00,0x06,0x01,0xD7}, 4, 0},
    {0x2A, (uint8_t []){0x00,0x06,0x01,0xDD}, 4, 0},
    {0x2B, (uint8_t []){0x00,0x00,0x01,0xD1}, 4, 0},
    {0x11, (uint8_t []){0x00}, 0, 60},
    {0x29, (uint8_t []){0x00}, 0, 0},
};
// LVGL library = not thread-safe, so we will call LVGL APIs from different tasks and use a mutex to protect it
static _lock_t lvgl_api_lock;

static esp_lcd_panel_handle_t panel_handle = NULL;

static bool notify_lvgl_flush_ready(esp_lcd_panel_io_handle_t panel_io, esp_lcd_panel_io_event_data_t *edata, void *user_ctx)
[... 94 lines cut ...]
        time_till_next_ms = MIN(time_till_next_ms, LVGL_TASK_MAX_DELAY_MS);
        usleep(1000 * time_till_next_ms);
    }
}

void app_main(void) {
    lv_init();
    lvgl_display_init();

    ESP_LOGI(TAG, "Create LVGL task");
    xTaskCreate(lvgl_port_task, "LVGL", LVGL_TASK_STACK_SIZE, NULL, LVGL_TASK_PRIORITY, NULL);

    ESP_LOGI(TAG, "Lock & Create Spinner");
    // Lock the mutex due to the LVGL APIs are not thread-safe
    _lock_acquire(&lvgl_api_lock);

    /*Create a spinner*/
    lv_obj_t * spinner = lv_spinner_create(lv_screen_active());
    lv_obj_set_size(spinner, 300, 300);
    lv_obj_center(spinner);
    lv_spinner_set_anim_params(spinner, 10000, 200);

    _lock_release(&lvgl_api_lock);
}
`

VIEWESMART, 2025-05-20:
> Thank you for your reply. Actually there is a co5300 driver available here: https://components.espressif.com/components/kodediy/esp_lcd_co5300 and it works in lvgl 8.3. I can post you my code if you want. **I'm trying to migrate to lvgl 9 without success so far. Display turns on but colors are wront a buggy. Can you help me out here?**
> 
> [image]
> 
> Here is my code:
> 
> #include <stdio.h>
> #include <unistd.h>
> #include <sys/lock.h>
> #include <sys/param.h>
> #include "freertos/FreeRTOS.h"
> #include "freertos/task.h"
> #include "driver/gpio.h"
> #include "driver/spi_master.h"
> #include "esp_timer.h"
> #include "esp_lcd_panel_io.h"
> #include "esp_lcd_panel_vendor.h"
> #include "esp_lcd_panel_ops.h"
> #include "esp_err.h"
> #include "esp_log.h"
> #include "lvgl.h"
> #include "esp_lcd_co5300.h"
> #include "ui.h"
> #include "lv_examples.h"
> 
> #define LCD_HOST               (SPI2_HOST)
> #define PIN_NUM_LCD_CS         (GPIO_NUM_12)
> #define PIN_NUM_LCD_PCLK       (GPIO_NUM_10)
> #define PIN_NUM_LCD_DATA0      (GPIO_NUM_13)
> #define PIN_NUM_LCD_DATA1      (GPIO_NUM_11)
> #define PIN_NUM_LCD_DATA2      (GPIO_NUM_14)
> #define PIN_NUM_LCD_DATA3      (GPIO_NUM_9)
> #define PIN_NUM_LCD_RST        (GPIO_NUM_8)
> #define PIN_NUM_BK_LIGHT       (GPIO_NUM_17)
> #define LCD_H_RES              471
> #define LCD_V_RES              466
> #define LCD_BIT_PER_PIXEL      16
> #define LCD_BK_LIGHT_ON_LEVEL  1
> 
> #define LVGL_DRAW_BUF_LINES    20 // number of display lines in each draw buffer
> #define LVGL_TICK_PERIOD_MS    2
> #define LVGL_TASK_MAX_DELAY_MS 500
> #define LVGL_TASK_MIN_DELAY_MS 1000 / CONFIG_FREERTOS_HZ
> #define LVGL_TASK_STACK_SIZE   (4 * 1024)
> #define LVGL_TASK_PRIORITY     2
> 
> static const char *TAG = "main";
> 
> // Custom initalizations according to vendor's example - doesn't seem to make a difference
> static const co5300_lcd_init_cmd_t lcd_init_cmds[] = {
>     {0xFE, (uint8_t []){0x00}, 0, 0},
>     {0xC4, (uint8_t []){0x80}, 1, 0},
>     {0x3A, (uint8_t []){0x55}, 1, 0},
>     {0x35, (uint8_t []){0x00}, 0, 10},
>     {0x53, (uint8_t []){0x20}, 1, 10},
>     {0x51, (uint8_t []){0xFF}, 1, 10},
>     {0x63, (uint8_t []){0xFF}, 1, 10},
>     //{0x2A, (uint8_t []){0x00,0x06,0x01,0xD7}, 4, 0},
>     {0x2A, (uint8_t []){0x00,0x06,0x01,0xDD}, 4, 0},
>     {0x2B, (uint8_t []){0x00,0x00,0x01,0xD1}, 4, 0},
>     {0x11, (uint8_t []){0x00}, 0, 60},
>     {0x29, (uint8_t []){0x00}, 0, 0},
> };
> // LVGL library = not thread-safe, so we will call LVGL APIs from different tasks and use a mutex to protect it
> static _lock_t lvgl_api_lock;
> 
> static esp_lcd_panel_handle_t panel_handle = NULL;
> 
> static bool notify_lvgl_flush_ready(esp_lcd_panel_io_handle_t panel_io, esp_lcd_panel_io_event_data_t *edata, void *user_ctx)
> {
[... 96 lines cut ...]
> }
> 
> void app_main(void) {
>     lv_init();
>     lvgl_display_init();
> 
>     ESP_LOGI(TAG, "Create LVGL task");
>     xTaskCreate(lvgl_port_task, "LVGL", LVGL_TASK_STACK_SIZE, NULL, LVGL_TASK_PRIORITY, NULL);
> 
>     ESP_LOGI(TAG, "Lock & Create Spinner");
>     // Lock the mutex due to the LVGL APIs are not thread-safe
>     _lock_acquire(&lvgl_api_lock);
> 
>     /*Create a spinner*/
>     lv_obj_t * spinner = lv_spinner_create(lv_screen_active());
>     lv_obj_set_size(spinner, 300, 300);
>     lv_obj_center(spinner);
>     lv_spinner_set_anim_params(spinner, 10000, 200);
> 
>     _lock_release(&lvgl_api_lock);
> }
> ``

Yes, I know this driver, but it does not belong to the official Espressif driver library, so it cannot be called in the display_panel, so it is not used.
For the color problem you want to use lvgl v9, it should be that a certain parameter is not set properly. I need to configure it myself, and I will reply to you tomorrow, China time.

qwerkus-1, 2025-05-20:
Ok wating for your reply. Here is my idf_component.yml for lvgl 9 on your device: 
`
dependencies:
  ## Required IDF version
  idf:
    version: '>=4.1.0'
  lvgl/lvgl: '*'
  espressif/knob: '*'
  espressif/button: '*'
  kodediy/esp_lcd_co5300: '*'
  kodediy/esp_lcd_touch_cst820: '*'
`

VIEWESMART, 2025-05-21:
> Ok wating for your reply. Here is my idf_component.yml for lvgl 9 on your device, if you want to reproduce my code.
> 
> dependencies:
>   ## Required IDF version
>   idf:
>     version: '>=4.1.0'
>   lvgl/lvgl: '*'
>   espressif/knob: '*'
>   espressif/button: '*'
>   kodediy/esp_lcd_co5300: '*'
>   kodediy/esp_lcd_touch_cst820: '*'

Are you using idf? Oak, that's much easier to handle.

qwerkus-1, 2025-05-22:
Any update on this ? LVGL 9 would be really nice on this display.

VIEWESMART, 2025-05-22:
> Any update on this ? LVGL 9 would be really nice on this display.

Yes, I'm already working on it. It may be delayed. I'll give it to you tomorrow at the latest.

VIEWESMART, 2025-05-24:
> Any update on this ? LVGL 9 would be really nice on this display.

I have synced my latest content to GitHub, linking to: https://github.com/VIEWESMART/UEDX46460015-MD50ESP32-1.5inch-Touch-Knob-Display/tree/main/examples/ESP-IDF/1.5inch_lvgl_v9_port

There is no problem with the color, because of the time relationship in the coordinate alignment you need to refer to the v8 configuration rounder_cb adjustment, you can also refer to here: https://github.com/Lzw655/esp-bsp/blob/d99913fd6a186a842a81b1543e1b3b6f146f1462/components/esp_lvgl_port/src/lvgl9/esp_lvgl_port_disp.c#L403

We didn't have time to deal with this piece, and we will further improve it in the future.

qwerkus-1, 2025-05-24:
Thank you for your efforts. Color correct now but lvgl9 animations are still slow and glitchy - see picture. Also, why do you include HF_ws2812 - there is no led strip on this device. 

[image]

qwerkus-1, 2025-05-24:
Not sure I'm following what you're trying to do: if you want to use lvgl_port, it makes the code simpler, but there is no option for a callback function so you cannot implement rounders. If you want to implement rounds, you need to configure lvgl manually and you don't need lvgl port. The esb-bp link you sent me seems unnecessary complex: they call lvgl_port and than configure the display manually !?

VIEWESMART, 2025-06-19:
> Not sure I'm following what you're trying to do: why use lvgl_port instead of setting up lvgl directly ? Half the options are unnecessary for this device, and lvgl_port doesn't let you control callback functions (like rounder_cb) directly. Anyway, I fixed my code with the rounder_cb function you linked, and now it works, but it's super slow. **I checked your code, and it is also very slow.** Something must be wrong in the way lvgl loads and flushes the buffer. Here my v9 with co5300 driver, no lvgl port - idf_component file at the end.
> 
> //*
>  * Knob-Style Ebike Display
>  *
>  * HW: VIEWE UEDX46460015-MD50E Round Amoled Board
>  */
> 
> /*
>  * Knob-Style Ebike Display
>  *
>  * HW: VIEWE UEDX46460015-MD50E Round Amoled Board
>  */
> 
> #include <stdio.h>
> #include <unistd.h>
> #include <sys/lock.h>
> #include <sys/param.h>
> #include "freertos/FreeRTOS.h"
> #include "freertos/task.h"
> #include "driver/gpio.h"
> #include "driver/spi_master.h"
> #include "esp_timer.h"
> #include "esp_lcd_panel_io.h"
> #include "esp_lcd_panel_vendor.h"
> #include "esp_lcd_panel_ops.h"
> #include "esp_err.h"
> #include "esp_log.h"
> #include "lvgl.h"
> #include "iot_knob.h"
> #include "iot_button.h"
> #include "esp_lcd_co5300.h"
> //#include "ui.h"
> #include "lv_examples.h"
> // display configuration
> #define LCD_HOST               (SPI2_HOST)
> #define PIN_NUM_LCD_CS         (GPIO_NUM_12)
> #define PIN_NUM_LCD_PCLK       (GPIO_NUM_10)
> #define PIN_NUM_LCD_DATA0      (GPIO_NUM_13)
> #define PIN_NUM_LCD_DATA1      (GPIO_NUM_11)
> #define PIN_NUM_LCD_DATA2      (GPIO_NUM_14)
> #define PIN_NUM_LCD_DATA3      (GPIO_NUM_9)
> #define PIN_NUM_LCD_RST        (GPIO_NUM_8)
> #define PIN_NUM_BK_LIGHT       (GPIO_NUM_17)
> #define LCD_H_RES              471
> #define LCD_V_RES              466
> #define LCD_BIT_PER_PIXEL      16
> #define LCD_BK_LIGHT_ON_LEVEL  1
> // encoder and button configuration
> #define BSP_ENCODER_A         (GPIO_NUM_6)
> #define BSP_ENCODER_B         (GPIO_NUM_5)
> typedef enum {
>     BSP_BTN_PRESS = GPIO_NUM_0,
> } bsp_button_t;
> // touch disabled by default
> #define  CONFIG_LCD_TOUCH_ENABLED  0
> #if CONFIG_LCD_TOUCH_ENABLED
> #define TOUCH_HOST                (I2C_NUM_0)
> #define PIN_NUM_TOUCH_SCL         (GPIO_NUM_3)
> #define PIN_NUM_TOUCH_SDA         (GPIO_NUM_1)
> #define PIN_NUM_TOUCH_RST         (GPIO_NUM_2)
> #define PIN_NUM_TOUCH_INT         (GPIO_NUM_4)
> esp_lcd_touch_handle_t tp = NULL;
> #endif
> // lvgl configuration
> #define LVGL_BUFF_SIZE         (LCD_H_RES * 80)
> #define LVGL_TICK_PERIOD_MS    2
> #define LVGL_TASK_MAX_DELAY_MS 500
> #define LVGL_TASK_MIN_DELAY_MS 2
> #define LVGL_TASK_STACK_SIZE   (4 * 1024)
[... 174 lines cut ...]
>     ESP_LOGE(TAG, "Failed to create LVGL task");
>     }
> 
>     ESP_LOGI(TAG, "Lock & Create UI");
>     // Lock the mutex because LVGL APIs are not thread-safe
>     _lock_acquire(&lvgl_api_lock);
> 
>     // create_ui();
>     // Spinner demo
>     
>     lv_obj_t * spinner = lv_spinner_create(lv_screen_active());
>     lv_obj_set_size(spinner, 300, 300);
>     lv_obj_center(spinner);
>     lv_spinner_set_anim_params(spinner, 5000, 200);
>     
>     _lock_release(&lvgl_api_lock);
> }
> idf_component.yml file:
> 
> dependencies:
>   lvgl/lvgl: 9.2.0
>   kodediy/esp_lcd_co5300: '*'
>   kodediy/esp_lcd_touch_cst820: '*'

I'm very sorry, I keep forgetting to reply, we have adapted 9.3.0 version together with lvgl, you can go and have a look: https://github.com/VIEWESMART/lv_port_viewe_knob_15_espidf/tree/main/components

xagon0, 2026-01-08:
On the most recent version, I was still running into the display corruption; but after extensive debugging, I found that the corruption when using LVGL 9 was that the SH8601 driver requires all coordinates to be even-aligned (divisible by 2), but LVGL 9's partial rendering sends odd-aligned coordinates which appears to corrupt the display.

If you add a counter callback, that forces all invalidated areas to have even-boundaries, then it works. I added this function before `app_lvgl_init()`:

```
  /**
   * Rounder callback to ensure coordinates are even-aligned.
   * SH8601 requires x_start, y_start, x_end, y_end to be divisible by 2.
   */
  static void lvgl_rounder_callback(lv_event_t *e)
  {
      lv_area_t *area = lv_event_get_param(e);

      // Round start coordinates down to nearest even number
      area->x1 = (area->x1 >> 1) << 1;
      area->y1 = (area->y1 >> 1) << 1;

      // Round end coordinates up to nearest odd number (so width/height are even)
      area->x2 = ((area->x2 >> 1) << 1) + 1;
      area->y2 = ((area->y2 >> 1) << 1) + 1;
  }
```

Then registered it after `lvgl_port_add_disp()` in `app_lvgl_initi()`:

```
lvgl_disp = lvgl_port_add_disp(&disp_cfg);

  /* Register rounder callback for SH8601 even-alignment requirement */
  lv_display_add_event_cb(lvgl_disp, lvgl_rounder_callback, LV_EVENT_INVALIDATE_AREA, NULL);
```

My guess here is that since LVGL9 uses partial rendering by default, sending small rectangular areas to update - when the driver gets non evenly-aligned coordinates, it corrupts.